phil-barrett/PicoCNC #5: U11 magic smoke and fire, Pi Pico W dead

iLoveAndyBaker, 2023-08-21:
U11 smoked, quite dramatically. 

This is the second Pi Pico that this PicoCNC killed. 

[image]

I suspect power issues on the board - , I haven't looked at the schematic however.  There was power cycling, I think it was probably my 12V (checked to 12.1V) supply resetting for an over-current situation until the chip got hot enough to go incandescent and stink up the whole room.  Thank god for the FR in FR4. 

Please double check your power

phil-barrett, 2023-08-21:
Wow.  That's pretty dramatic.  I've been testing a number of these boards running on 12V power for a lot of hours and never had anything blow at all.  That device (74LVC1G32) runs on +5V and the pin that is burnt is the +5V pin.  Bad chip?  Maybe 12V got onto the 5V rail?  Could you show me a picture of the entire board?

Please return the board and we will refund your purchase.

iLoveAndyBaker, 2023-08-21:
[image]
[image]
[image]

Let me know where to mail it and I'll get it out soon.

phil-barrett, 2023-08-21:
Phil Barrett
76 Olympic Dr NW
Shoreline, WA 98177

phil-barrett, 2023-08-22:
I have been looking at the design for how this could happen and I really can't see anyway for that part to go short unless it was a bad part.  This circuit is in two designs with over 1100 units in the field and no reports of a problem.  I see several possibilities - 1) bad chip with an internal short (very rare), 2) shorting across the output (not obvious from the photos) or 3) chip damaged due to 12V across the 5V rail (hard to do accidentally).

iLoveAndyBaker, 2023-08-22:
Nothing, huh? Yeah I don't know either. Maybe something etched weird. Maybe a part was just bad, or wrong, or just randomly latched up. Either way, it's in the mail, I've thrown in the blown Pico as a bonus. I hope you can learn something from it, it looks like a nice little board. I would have liked to have put the FLIR camera on it, but as luck would have it, I've lent it out.

Thanks,

Andy

phil-barrett, 2023-08-30:
I got the board and did some poking. Powered it up with 12V.  The first thing that became clear is there was 11V on the 5V rail.  That pretty much blew all the 5V logic chips and the Pico.  I'd guess U11 being so small, it went first.  Looking at the 5V regulator U1 (AMS1117-5), I could see some damage to it. See the photo. (I made the scratch probing at it.) It failed with a direct short which is unusual - they usually die open. Maybe it took too much current and blew. Maybe too high voltage was applied - 24V would have killed it. Maybe reversed polarity? I looked at all the boards I have in stock (55) and none of them showed any damage to the regulator.  I desoldered the dead regulator and soldered a new one in place.  It happily produced 5V and did not overheat.  I did not see any evidence of over current any other place than U1 and U11.  So, it's hard to say what damaged U1.  I desoldered U11 and inspected the traces under it.  They appear to be ok so it wasn't a board short.  I also looked at the pin headers used to join the 2 12V sections to see if the shorting plugs had been installed wrong but did not see any damage.   

[image]

So what did I learn from this?  To be honest, not a lot. A lot of this circuitry is the same as on the Teensy board. And the circuit is pretty straightforward. There is no obvious design flaw. I am thinking of adding an in line rectifier before the 12V input of the AMS1117 to at least protect it from reverse polarity though that has implications on the 0-10V circuit.  I had already separated the headers for joining the 2 12V sections to prevent accidental shorting (that's in V1.51, the currently shipping board). I am adding warnings to the User Manual about reverse polarity and power sources above 12V.

Anyway, I believe this is an isolated case.  There have been no other reports of a similar nature and in all the testing I never saw any significant overheating (I do a lot of thermal measurement on the boards during testing). In general, the worst case temperature increase is 15C over ambient, well within expectations.

Anyway, I have issued a full refund. Sorry this happened.
Phil Barrett

iLoveAndyBaker, 2023-08-30:
I'd be happy to take a new one to be honest. It makes my wiring a lot cleaner too.

The power supply was a nice switching wall wart putting out clean 12.1V.  It was running well, I connected it when I started the project and never looked back. As to why I was thinking those LEDs too bright, maybe it was that regulator after all. At any rate, I can use a different PSU.

I've seen people put big fat diodes across the input.. blows the fuse in the PSU if you go backwards, and no voltage drop that way. Provided it's got a fuse that is.

Thanks for looking into this.

Andy Baker
Red 6

[dead board]<https://user-images.githubusercontent.com/16570958/264234208-1ab6f625-d3bf-439e-bbfd-191e6f0cb7cf.jpg>

phil-barrett, 2023-08-30:
With the refund, you could place another order. Tindie is prompt about refunds.

phil-barrett, 2023-09-02:
I'm closing this. There have been no more reports of anything similar so I am calling a one-off problem. If anything comes up I will reopen.

closed 2023-09-02 by phil-barrett