phil-barrett/PicoCNC #9: Brake Out Board for Blackpill (STM32F411CE - WeAct Studio)

wolfgangbregel, 2023-11-02:
Hello Phill,

I much appreciate your work and the well designed BOBs for Pico and Teensy.

Just a proposal: Why not consider the low cost Blackpill (STM32F411CE - WeAct Studio) as a controller base?

* Well supoported by GRBL HAL (Flawless compile with STM32Cube Ide - or GRBL Web builder)
* Just enough GPIOs (a total of 32 +2 used for SW-JTAG)  for a 4 or 5 axis CNC including per axis limit switches and inputs for Safety Door, Feed Hold, Cycle Start, Reset E-Stop and Probe
* Outputs for Spindle (Dir, Enable, PWM), Coolant Flood and Mist
* Optional inputs for Spindle Pulse / Index
* No need for external serial->parallel Expander chips at high speed clocks
* Integrated USB CDC
* Works just fine on my CNC metal milling machine (On my home brew breakout board for now)
* Plenty of headroom in Ram amd Flash for extensions and future development

Many thanks for considering,

Wolf

phil-barrett, 2023-11-02:
Thanks for the input.  As usual, there are always trade-offs to be taken into account. Such is engineering...

I've considered the Black Pill before and it would be a very reasonable choice on many fronts.  One problem is that Chinese suppliers often change components without any notice.  This includes the actual processor (there are a number of F411 clones out there that MOSTLY work the same as the STM part).  Also, the availability of the Black Pill is highly variable. I have looked at using the F411 directly on a base board but the chip shortage hit the STM32 line very hard. The RP2040 never had that problem and it is a consistent product, never any silent changes. As we emerge from the still lingering effects of the shortage, I will continue to re-evaluate things.

The main difference is that the PicoCNC board only supports 4 Axes. The PicoCNC step output rate is well above 200 KHz, supports WiFi, doesn't need a programmer and is super inexpensive.  Not suitable for lathes, though. And, soon I will release a PicoCNC that will support Ethernet.  A Blackpill based board with Ethernet would probably not be able to support 5 Axes in that case.

wolfgangbregel, 2023-11-02:
Hello Phil,

Yes, I fully understand your concerns as to chinese suppliers and their
liberty in changing specs and/or chips.
And yes the Pico is well documented and more consistent over time since its
appearance.
I want to replace an Arduino based control board (i.e. no interference
protection - but no EMC problems either) in order to have a 4th axis and
backlash compensation.
So - I'll order one of your PicoCNC boars shortly. I just had a few really
annoying surprices from the German customs dept. I.e. calling for next to
half of the invoice as import tax for electronics received from the US.

Greatings and thumbs up from Germany

Wolfgang

phil-barrett, 2023-11-02:
Thanks!  I have a lot of German customers, it is my second biggest market (after USA), and have not heard them complain about getting nicked for customs/duty/VAT. I do supply correct documentation including price so the max should be a 20% charge, I think. (I will admit, it's hard to keep track of it.) Maybe they just don't tell me. Note, I use the US postal system which hands off to Deutsche Post and not carriers like FedEx or UPS. The few times I have used UPS internationally they not only hit the customer for Duty/VAT and ignored the listed price but also added a $30 or so brokerage fee. One Irish customer was charged $70 on a $70 order. Thieves.

wolfgangbregel, 2023-11-03:
Hello Phil,

many thanks for your info. Indeed, Deutsche Post/DHL is the most reliable,
honest and mostly cheapest service here in Germany.
Good to hear that US postals works with DHL only.
I once had an experience with UPS for an order I accidently placed with
DigiKey US. I had to pay about $20 service fee on the $50 order - heavy
(plus VAT of course).

So i checked for orders from the US to Germany *below *€150 (*above *is
different):

Duty: None
Import VAT: 19% (on the invoice total, incl. transport!)
DHL service fee: €6

Just for info, in case someone asks.

I'll order a PicoCNC via Tindie shortly.

Many thanks for the great job!

Best regrards, Wolfgang

wolfgangbregel, 2023-11-03:
Hello Phil,

just to let you know: I just ordered a PicoCNC assy kit via Tinide.

Best regards, Wolfgang

phil-barrett, 2023-11-03:
Thank You.  

You should have email confirmation and the package will ship in about 1 hour.

Phil

wolfgangbregel, 2023-11-03:
Hello Phil,

yes, just received a confirmation from US postal.

Many thanks, Wolfgang

closed 2024-02-16 by phil-barrett