phil-barrett/PicoCNC #12: Best way to buy from Europe (Italy)

yaapu, 2024-01-03:
Hi Phil,
I'd like to get one of your PicoCNC boards, I tried tindie but the only shipping option is "United States Postal Service Standard Ground Rate (No Tracking)", I'd rather use a tracked service since custom duties are involved, do you have any hints/ideas?

phil-barrett, 2024-01-03:
Actually, you will get tracking with that service.  I should revisit the tindie shipping setup.

phil-barrett, 2024-01-03:
Fixed it.  Now it says tracking.  

Note that this is a US Postal Service tracking number and may not give you tacking within Italy.  I have seen cases in a number of countries where tracking is not updated after the hand over to the local postal service.  And, I have seen cases where a given country's tracking is inconsistent - sometimes they do it but other times they don't.  Very annoying and there is no recourse.  Sorry, I wish it was better.

Phil

yaapu, 2024-01-03:
Thanks for your super quick reply, I suspect that international orders will not use Ground Service, do you have an idea of which tracked service will be used?
Any track record of how long it would take to get it to italy or europe?

phil-barrett, 2024-01-03:
USPS hands the package off to whatever the designated carrier is for the end country.  Usually it is the national mail carrier but I don't have any visibility into that.

Ship time to EU countries can vary a lot.  1 week to 4 weeks. The last several I shipped to Italy ranged between 13 and 17 days. Once the package gets to Italy, it appears to take just a few days.  The biggest delay for international shipping is the gap between the last US reported location and the first end country location.  That is between 3 and 10 days, typically.

yaapu, 2024-01-03:
Thanks Phil

closed 2024-01-03 by yaapu