phil-barrett/PicoCNC #16: Ethernet + sdcard + i2c panel

pimsierhuis, 2024-02-10:
We like to use ethernet, sd card and use an i2c board, so we're running into the error "I2C strobe pin not available, is assigned as ethernet IRQ!". Could there be any chance of supporting that configuration? I think it could be a quite useful use case.

phil-barrett, 2024-02-10:
I totally agree - a very useful case. Unfortunately, the Pico just doesn't have enough pins to support both. I am working towards supporting that scenario on the next version of the PicoCNC as it will use the RP2040 directly and get a few more pins - I2C and Ethernet concurrent are absolutely the top priority.  We are at least 6 months from that being available.

An alternative is to use the Teensy version which does support Ethernet, SDCard and I2C concurrently.

Another possibility if you are designing the keypad HW is to use the UART.

Phil

pimsierhuis, 2024-02-11:
OK, thanks for your explanation!

Will stick with a panel via UART for now then, can always upgrade later.

closed 2024-02-16 by phil-barrett