phil-barrett/PicoCNC #17: Ethernet on v1.52

JelmerV, 2024-02-12:
Hi, I bought 2 PicoCNC boards just before the update that added the Wiz850io header. However, I realize Ethernet will be very useful and would like to add it.

Do you have any advice on wiring it and configuring the Wiz850io on a PicoCNC V1.52?

phil-barrett, 2024-02-12:
It is possible but messy.  You would need cut a few traces and add 3 bodge wires.  In addition, you would need a way to secure the Wiz850io.  I would be willing to give you a 20% discount on replacement boards/kits if you would prefer. Shipping cost will be better, too. Talk to me directly, not through Tindie.

If you want to go the bodge wire approach, let me know and I will write something up.

JelmerV, 2024-02-12:
Thanks for the offer! However for the time being I would prefer to work with the older boards that I've got. If you could just share which connections to make that would be enough. I have some additional circuitry to add on perfboard anyway so securing it shouldn't be a problem.

phil-barrett, 2024-02-12:
OK.   I will write it up in the next day or so.  I may do a blog post.  Look in the discussions section for either the write up or a pointer to the blog post.

phil-barrett, 2024-02-16:
Posted on Discussions.

closed 2024-02-16 by phil-barrett