phil-barrett/PicoCNC #29: Using a lower voltage for the opto-isolated input

BenjaminPoilve, 2024-11-11:
I plan on using optical endstops on my machine, and they rely on pre-built board using 5V as signal level. 
Am I right to think that I could just replace the 12V ISO input by a 5V power supply, and get on with it (with of course a significant reduction in EMI resistance)?
The led in the opto-isolator would get a bit less than 2mA, which is not great, but should be enough to pull the logic input down, and if not, I'd just have to replace the led resistors. 

Is there anything else that I should keep in mind for that modification? 

Thanks in advance.

phil-barrett, 2024-11-11:
I have not tested the PicoCNC with 5V iso voltage.  The resistors on the Opto Isolators are 2.2K.  With a nominal voltage drop of 1.2V, that leaves 1.7 mA flowing through the EL3H7 opto.  While this is not ideal, the datasheet suggests that it can work.  If you have problems (like EMI susceptibility), I would replace the resistors with 470 ohm (603 SMT) which would yield 8 mA opto current and a much more robust current transfer ratio. I can talk you through that if you need.

Your board went out in the morning mail drop to the post office, by the way.

phil-barrett, 2025-12-07:
I have released all the design files.  Please feel free to make changes and post back your progress!