phil-barrett/PicoCNC #30: Pico 2 support?

Dexus, 2024-12-01:
Would this word support the pico2? Or only the pico1?

phil-barrett, 2024-12-01:
Yes, the Pico2 fits in the board.  Currently, there is not a version of grblHAL that supports it but I have heard that it may be coming.

Phil

phil-barrett, 2024-12-04:
We are making progress.  The Web Builder now supports building grblHAL for a Pico2 (RP2350) plugged into a PicoCNC board.  There are a couple of issues with timing for step generation that I think will be fixed soon. 

[image]

Dexus, 2024-12-04:
Have seen it. Now I await my order be shipped ;) from US to DE. Thank you.

one more thing I would like to know it should be easy to use a diode laser on that board right? Or should I know something special?

Best Regards josef

phil-barrett, 2024-12-04:
[This might help](https://www.grbl.org/single-post/using-a-diode-laser-with-grblhal)

Dexus, 2025-01-17:
@phil-barrett only to be sure, i have to switch the 0 resistor from R40 to R39 right for laser mode?